tum-ei-eda/M2-ISA-R #5: Nested static statements

fabianpedd, 2021-09-02:
M2-ISA-R currently produces "nested" static statements by calling `make_static()` on statements that were already made static. This leads to code like this:

```
partInit.code() += "_illmask = " + std::to_string((1) << (" + std::to_string(32 - 1) + ")) + ";\n";
```

See 16e0bcb120a147d9375efd2e8368f67256114e81 for an extremely dirty fix.

wysiwyng, 2021-09-03:
That fix is indeed extremely dirty. Do these nested static statements only happen with the builtins `shrl`, `shll` and `shra`? These functions currently don't support static expressions as their first argument. As a workaround, use the standard shifting operators `<<` and `>>`.

fabianpedd, 2021-09-03:
I have not tested anything beyond the vector CoreDSL. In there the only cases this happened seemed to be with the builtins. I will give the standard shifts a try. 

If I can simply use the standard shifts then what is the reason behind the builtins anyways?

wysiwyng, 2021-09-03:
The shift builtins allow explicit arithmetic or logical shifts. Apart from that, they do not provide any different functionality. They will be removed in CoreDSL 2.0, where only the standard shift operators are provided.

wysiwyng, 2021-09-05:
I pushed a fix for this, please test and let me know whether this indeed fixes the problem.

fabianpedd, 2021-09-06:
cfd3b6e93bacad47b343235ee75cd798096aad74 and b9579babcc0b3527d4c55589061e7e4b017c31f1 appear to have fixed the issue. However, I would leave this issue open until I have conducted some tests.

fabianpedd, 2021-11-14:
Sorry for the late reply. I completely overlooked this. I haven't been using M2-ISA-R very much lately as I need to implement some more RVV instructions in order to use ETISS for my work. I am waiting for the M2-ISA-R support for CoreDSL 2 in order to avoid having to reimplement (or at least refactor) everything. Thus, I would close this issue for now and come back to it if I run into issues again.

closed 2021-11-14 by fabianpedd