tum-ei-eda/M2-ISA-R #10: Overflow dropping in summation operation

togulcan, 2022-03-21:
I realized during the below operation wrong return value `Z ` is generated during the summation of X and Y:
```
unsigned<64> foo(unsigned<32> X, unsigned<32> Y){
    unsigned<64> Z = X + Y;
    return Z;
}
```
I believe this is due to the fact that `X ` and `Y ` are translating to C types (`etiss_uint32`) and their summation also returns `etiss_uint32` which drops the overflow value(32nd value).

However when I do casting explicitly as follows:
```
unsigned<64> foo(unsigned<32> X, unsigned<32> Y){
    unsigned<64> Z = (unsigned<64>)X + (unsigned<64>)Y;
    return Z;
}
```
It works pretty well.

But [coredsl2 manual](https://github.com/Minres/CoreDSL/wiki/Expressions#addition) guarantees no dropping is occurring by simply setting the width of there result value to `wr = max(w1 + 1, w2 + 1)`. Should not M2-ISA-R also guarantee this? This is leading to a lot of bugs.

Thanks a lot!

neithernut, 2022-03-21:
AFAIU this is a bug.

wysiwyng, 2022-03-21:
This is indeed an oversight in the implementation of at least the ETISS code generator. CoreDSL type promotion rules are more or less ignored completely in M2-ISA-R as of now, therefore standard C type promotion rules are used in ETISS. I will look into this, but it will probably take a moment to properly fix.

PhilippvK, 2025-03-02:
TODO: check if still relevant