tum-ei-eda/M2-ISA-R #12: Sliced width is ignored during concatenation

togulcan, 2022-03-24:
Hello,

the following coredsl2 code:
```
void foo(){
  unsigned<8> a = 3;
  unsigned<8> b = 1;
  unsigned<32> x = 0, y=0;
  unsigned<64> z = x::y[a:b];
}
```
is producing:
```
static inline void foo (){
  etiss_uint8 a = 3U;
  etiss_uint8 b = 1U;
  etiss_uint32 x = 0U;
  etiss_uint32 y = 0U;
  etiss_uint64 z = (((x) << 32) | ((((y) >> (b)) & ((1 << ((a) - (b) + 1)) - 1))));
}
```
where x has been shifted `32` times instead of `a - b + 1` times.

However, putting numbers into slices:
```
void foo(){
  unsigned<32> x = 0, y=0;
  unsigned<64> z = x::y[3:1];
}
```
works quite well:
```
static inline void foo (){
  etiss_uint32 x = 0U;
  etiss_uint32 y = 0U;
  etiss_uint64 z = (((x) << 3) | ((((y) >> (1U)) & 7)));
}
```

any idea about this?

wysiwyng, 2022-07-15:
This would be a bug in M2-ISA-R and is on the roadmap to be supported, but at the time being, slicing is only supported with constants, see here: https://github.com/Minres/CoreDSL/wiki/Expressions#bit-range. I'll leave this open, as theoretically supporting non-constant slices should not be a problem for M2-ISA-R and ETISS.

PhilippvK, 2023-05-23:
@wysiwyng I guess this can be closed?