tum-ei-eda/M2-ISA-R #16: PC access in functions does not yield correct results

wysiwyng, 2022-06-03:


neithernut, 2022-06-03:
Could you please elaborate?

wysiwyng, 2022-06-03:
the PC can be read as `ic.current_address_` during the instruction translation phase of ETISS, as functions are executed during JIT runtime they need to read the PC as `cpu->instructionPointer`

wysiwyng, 2022-06-03:
problematic in "just doing that" is that currently `cpu->instructionPointer` is incremented at the very beginning of each instruction, but `ic.current_address_` always points to the address of the current instruction. to fix this properly, the PC increment would have to be reworked, and by that extent most likely made explicit in the CoreDSL model.

wysiwyng, 2022-07-15:
see also: https://github.com/Minres/CoreDSL/issues/54

PhilippvK, 2025-03-02:
TODO: check if still relevant