tum-ei-eda/M2-ISA-R #26: m2isar.backends.etiss.instruction_writer: Generatation of literals > 64bit

PhilippvK, 2024-04-04:
C does only provide literal suffices for up to 64 bit large constants (`ULL`).

We sometimes need bitmasks which are 65 bits or larger: `0x1ffffffffffffffff`

The only workaround I have in mind for this is building the literal in an `etiss_int128 temp` variable manually...

wysiwyng, 2024-04-04:
We could add generation code that splits a > 64 bit literal into two 64 bit literals and combines them back together. Alternatively, a hacky solution would be this: https://stackoverflow.com/a/33331273.

wysiwyng, 2024-04-04:
In C++ this exists: https://en.cppreference.com/w/cpp/language/user_literal. Irrelevant for us unfortunately as the JIT is only C.